YJZLuckyBoy/liorf #3: Does this work with the internal IMU of OS1-128?

PerOveL, 2022-12-19:
Hi. Great addition to LIO_SAM!

Is the 6DOF support for using the internal IMU of the Ouster OS1-128?

Best regards 
Per-Ove

YJZLuckyBoy, 2022-12-19:
Thanks for your attention to liorf. Theoretically, liorf is easy to work with the internal IMU of OS1-128, like the internal IMU of Livox, but I haven't tested it yet.

jeychandar, 2023-11-06:
@YJZLuckyBoy Hi Sir. I have tested with os1-128 it gives large velocity reset IMU-Preintegration eventhough I reduced the playback speed to -r 0.1 it gives the same error. Could you help me out with this issue. Thanks in advance

YJZLuckyBoy, 2023-11-08:
> @YJZLuckyBoy Hi Sir. I have tested with os1-128 it gives large velocity reset IMU-Preintegration eventhough I reduced the playback speed to -r 0.1 it gives the same error. Could you help me out with this issue. Thanks in advance

Can you provide some data?

CantLooseToAMoose, 2024-06-05:
> @YJZLuckyBoy Hi Sir. I have tested with os1-128 it gives large velocity reset IMU-Preintegration eventhough I reduced the playback speed to -r 0.1 it gives the same error. Could you help me out with this issue. Thanks in advance

Have you found a solution for this?