YJZLuckyBoy/liorf #4: Compiling erros

FPSychotic, 2022-12-25:
Hi, thanks by share this, I found a compiling error due to set(CMAKE_CXX_FLAGS "-std=c++11")
soved with set(CMAKE_CXX_FLAGS "-std=c++14")

later I got a CV error:
``[ 71%] Built target liorf_generate_messages
[ 73%] Building CXX object livox_ros_driver/livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lds_lidar.cpp.o
[ 75%] Building CXX object livox_ros_driver/livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lds_hub.cpp.o
[ 75%] Built target livox_ros_driver_generate_messages_eus
[ 77%] Building CXX object livox_ros_driver/livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/lddc.cpp.o
[ 80%] Building CXX object livox_ros_driver/livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/livox_ros_driver/livox_ros_driver.cpp.o
In file included from /home/iscan/lio_sam_ws/src/liorf/src/imuPreintegration.cpp:1:
/home/iscan/lio_sam_ws/src/liorf/include/utility.h:19:10: fatal error: opencv/cv.h: No such file or directory
   19 | #include <opencv/cv.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [liorf/CMakeFiles/liorf_imuPreintegration.dir/build.make:63: liorf/CMakeFiles/liorf_imuPreintegration.dir/src/imuPreintegration.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1140: liorf/CMakeFiles/liorf_imuPreintegration.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 82%] Building CXX object livox_ros_driver/livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/timesync/timesync.cpp.o
[ 84%] Building CXX object livox_ros_driver/livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/timesync/user_uart/user_uart.cpp.o
[ 86%] Building CXX object livox_ros_driver/livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/common/comm/comm_protocol.cpp.o
In file included from /home/iscan/lio_sam_ws/src/liorf/src/imageProjection.cpp:1:
/home/iscan/lio_sam_ws/src/liorf/include/utility.h:19:10: fatal error: opencv/cv.h: No such file or directory
   19 | #include <opencv/cv.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [liorf/CMakeFiles/liorf_imageProjection.dir/build.make:63: liorf/CMakeFiles/liorf_imageProjection.dir/src/imageProjection.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:694: liorf/CMakeFiles/liorf_imageProjection.dir/all] Error 2
[ 88%] Building CXX object livox_ros_driver/livox_ros_driver/CMakeFiles/livox_ros_driver_node.dir/common/comm/sdk_protocol.cpp.o
In file included from /home/iscan/lio_sam_ws/src/liorf/src/mapOptmization.cpp:1:
/home/iscan/lio_sam_ws/src/liorf/include/utility.h:19:10: fatal error: opencv/cv.h: No such file or directory
   19 | #include <opencv/cv.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [liorf/CMakeFiles/liorf_mapOptmization.dir/build.make:63: liorf/CMakeFiles/liorf_mapOptmization.dir/src/mapOptmization.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:606: liorf/CMakeFiles/liorf_mapOptmization.dir/all] Error 2

``
I could compile lio-sam without issues. 20.04, Noetic, stock opencv 4.2

thanks!

YJZLuckyBoy, 2022-12-27:
Thanks for your advice! 
I have fixed this bug, you can try it again.
I have tested with 18.04，but have not tested with 20.04 yet.  
For more information, please refer to [Issues #206](https://github.com/TixiaoShan/LIO-SAM/issues/206)

FPSychotic, 2022-12-27:
Hi, I will test it for you now then and report today, thanks by your fast response.

FPSychotic, 2022-12-27:
Just confirm you it will compile and work.
What I found, not sure if it is important for you, it wont appear as a ros pkg with roslaunch, is needed put manual the full address, in my case as follow.
roslaunch lio_sam_ws/src/liorf/launch/run_lio_sam_livox.launch 

my other problem no related is how to setup the config file for livox avia and its internal imu, in my case probably the default data of noise and bias are not enough good by default, if by causality you use a Livox Avia too would be nice if you can share the setup

FPSychotic, 2022-12-27:
I got this graph, I don't know if it is possible extract the data from it.
[image]
in any case maybe helps someone, it is a BMI088

YJZLuckyBoy, 2023-01-11:
> Just confirm you it will compile and work. What I found, not sure if it is important for you, it wont appear as a ros pkg with roslaunch, is needed put manual the full address, in my case as follow. [...]

Sorry for the late reply！ For the built-in imu of livox, the noise parameters in my yaml are as follows:
```
  imuAccNoise: 0.00059
  imuGyrNoise: 0.000061
  imuAccBiasN: 0.000011
  imuGyrBiasN: 0.000001
```
By the way, liorf can not directly use Livox lidar for mapping. You need to use a specific driver([livox_ros_driver](https://github.com/TixiaoShan/livox_ros_driver)  modified by TixiaoShan) or use a custom data conversion node.

closed 2023-12-13 by YJZLuckyBoy