YJZLuckyBoy/liorf #6: Drifting on MulRan Datasets

v3loo, 2023-01-11:
Thanks for your great work.

I recently tested this solution on MulRan dataset, but got drifting on every subset after running for couple seconds. It warns about 'large velocity, reset IMU-preintegration'. This happened both with and without GPS data.

I've set lidar type to MulRan as suggested, set corresponding extrinsics for lidar to imu, set number of scan to 64, and also set message topics correctly. Others are just kept the same as the default of 'lio_sam_ouster.yaml'.

Since you have mentioned your result on MulRan dataset, would you share your .yaml file for MulRan?

YJZLuckyBoy, 2023-01-11:
You can try again with the following parameters：

liorf:

  # Topics
  pointCloudTopic: "/os1_points"               # Point cloud data
  imuTopic: "/imu/data_raw"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useImuHeadingInitialization: true           # if using GPS data, set to "true"
  useGpsElevation: false                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 25.0                      # m^2, threshold for using GPS data
  
  # Export settings
  savePCD: false                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "/Downloads/LOAM/"        # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

  # Sensor Settings
  sensor: mulran                            # lidar sensor type, 'velodyne' or 'ouster' or 'livox' or 'robosense'
  N_SCAN: 64                                  # number of lidar channel (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
  Horizon_SCAN: 1024                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
  downsampleRate: 2                           # default: 1. Downsample your data if too many points(line). i.e., 16 = 64 / 4, 16 = 16 / 1
  point_filter_num: 5                         # default: 3. Downsample your data if too many points(point). e.g., 16: 1, 32: 5, 64: 8
  lidarMinRange: 3.0                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 1000.0                       # default: 1000.0, maximum lidar range to be used

  # IMU Settings
  imuType: 0                                  # 0: 6-axis  1: 9-axis
  imuRate: 100.0
  imuAccNoise: 0.009939570888238808e-03
  imuGyrNoise: 0.005636343949698187e-03
  imuAccBiasN: 0.64356659353532566e-03
  imuGyrBiasN: 0.35640318696367613e-03
  imuGravity: 9.80511
  imuRPYWeight: 0.01

  # Extrinsics: T_lb (lidar -> imu)
  extrinsicTrans: [1.77, -0.00, -0.05]
  extrinsicRot: [-1, 0, 0,
                  0, -1, 0,
                  0, 0, 1]

  # This parameter is set only when the 9-axis IMU is used, but it must be a high-precision IMU. e.g. MTI-680
  extrinsicRPY: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]

  # voxel filter paprams
  mappingSurfLeafSize: 0.4                      # default: 0.4 - outdoor, 0.2 - indoor

  # robot motion constraint (in case you are using a 2D robot)
  z_tollerance: 1000                            # meters
  rotation_tollerance: 1000                     # radians

  # CPU Params
  numberOfCores: 4                              # number of cores for mapping optimization
  mappingProcessInterval: 0.0                  # seconds, regulate mapping frequency

  # Surrounding map
  surroundingkeyframeAddingDistThreshold: 1.0   # meters, regulate keyframe adding threshold
  surroundingkeyframeAddingAngleThreshold: 0.2  # radians, regulate keyframe adding threshold
  surroundingKeyframeDensity: 2.0               # meters, downsample surrounding keyframe poses   
  surroundingKeyframeSearchRadius: 50.0         # meters, within n meters scan-to-map optimization (when loop closure disabled)
  surroundingKeyframeMapLeafSize: 0.5           # downsample local map point cloud

  # Loop closure
  loopClosureEnableFlag: true
  loopClosureFrequency: 1.0                     # Hz, regulate loop closure constraint add frequency
  surroundingKeyframeSize: 50                   # submap size (when loop closure enabled)
  historyKeyframeSearchRadius: 15.0             # meters, key frame that is within n meters from current pose will be considerd for loop closure
  historyKeyframeSearchTimeDiff: 30.0           # seconds, key frame that is n seconds older will be considered for loop closure
  historyKeyframeSearchNum: 25                  # number of hostory key frames will be fused into a submap for loop closure
  loopClosureICPSurfLeafSize: 0.5               # downsample icp point cloud  
  historyKeyframeFitnessScore: 0.3              # icp threshold, the smaller the better alignment

  # Visualization
  globalMapVisualizationSearchRadius: 1000.0    # meters, global map visualization radius
  globalMapVisualizationPoseDensity: 10.0       # meters, global map visualization keyframe density
  globalMapVisualizationLeafSize: 1.0           # meters, global map visualization cloud density

YJZLuckyBoy, 2023-01-11:
You can try again with the following parameters：
```
liorf:

  # Topics
  pointCloudTopic: "/os1_points"               # Point cloud data
  imuTopic: "/imu/data_raw"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useImuHeadingInitialization: true           # if using GPS data, set to "true"
  useGpsElevation: false                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 25.0                      # m^2, threshold for using GPS data
  
  # Export settings
  savePCD: false                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "/Downloads/LOAM/"        # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

  # Sensor Settings
  sensor: mulran                            # lidar sensor type, 'velodyne' or 'ouster' or 'livox' or 'robosense'
  N_SCAN: 64                                  # number of lidar channel (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
  Horizon_SCAN: 1024                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
  downsampleRate: 2                           # default: 1. Downsample your data if too many points(line). i.e., 16 = 64 / 4, 16 = 16 / 1
  point_filter_num: 5                         # default: 3. Downsample your data if too many points(point). e.g., 16: 1, 32: 5, 64: 8
  lidarMinRange: 3.0                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 1000.0                       # default: 1000.0, maximum lidar range to be used

  # IMU Settings
  imuType: 0                                  # 0: 6-axis  1: 9-axis
  imuRate: 100.0
  imuAccNoise: 0.009939570888238808e-03
  imuGyrNoise: 0.005636343949698187e-03
  imuAccBiasN: 0.64356659353532566e-03
  imuGyrBiasN: 0.35640318696367613e-03
  imuGravity: 9.80511
  imuRPYWeight: 0.01

  # Extrinsics: T_lb (lidar -> imu)
  extrinsicTrans: [1.77, -0.00, -0.05]
  extrinsicRot: [-1, 0, 0,
                  0, -1, 0,
                  0, 0, 1]

  # This parameter is set only when the 9-axis IMU is used, but it must be a high-precision IMU. e.g. MTI-680
  extrinsicRPY: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]

  # voxel filter paprams
  mappingSurfLeafSize: 0.4                      # default: 0.4 - outdoor, 0.2 - indoor

  # robot motion constraint (in case you are using a 2D robot)
  z_tollerance: 1000                            # meters
  rotation_tollerance: 1000                     # radians

  # CPU Params
  numberOfCores: 4                              # number of cores for mapping optimization
  mappingProcessInterval: 0.0                  # seconds, regulate mapping frequency

  # Surrounding map
  surroundingkeyframeAddingDistThreshold: 1.0   # meters, regulate keyframe adding threshold
  surroundingkeyframeAddingAngleThreshold: 0.2  # radians, regulate keyframe adding threshold
  surroundingKeyframeDensity: 2.0               # meters, downsample surrounding keyframe poses   
  surroundingKeyframeSearchRadius: 50.0         # meters, within n meters scan-to-map optimization (when loop closure disabled)
  surroundingKeyframeMapLeafSize: 0.5           # downsample local map point cloud

  # Loop closure
  loopClosureEnableFlag: true
  loopClosureFrequency: 1.0                     # Hz, regulate loop closure constraint add frequency
  surroundingKeyframeSize: 50                   # submap size (when loop closure enabled)
  historyKeyframeSearchRadius: 15.0             # meters, key frame that is within n meters from current pose will be considerd for loop closure
  historyKeyframeSearchTimeDiff: 30.0           # seconds, key frame that is n seconds older will be considered for loop closure
  historyKeyframeSearchNum: 25                  # number of hostory key frames will be fused into a submap for loop closure
  loopClosureICPSurfLeafSize: 0.5               # downsample icp point cloud  
  historyKeyframeFitnessScore: 0.3              # icp threshold, the smaller the better alignment

  # Visualization
  globalMapVisualizationSearchRadius: 1000.0    # meters, global map visualization radius
  globalMapVisualizationPoseDensity: 10.0       # meters, global map visualization keyframe density
  globalMapVisualizationLeafSize: 1.0           # meters, global map visualization cloud density
```

v3loo, 2023-01-12:
Thanks for your reply.

I've tested on MulRan with this config but it didn't work. It always drifts soon after the first turning.

Do you have any ideas?

> You can try again with the following parameters：
>
> ```
> liorf:
>
>   # Topics
>   pointCloudTopic: "/os1_points"               # Point cloud data
>   imuTopic: "/imu/data_raw"                         # IMU [...]

YJZLuckyBoy, 2023-01-13:
For the highway scene, you need to reduce the point cloud sampling rate, you can try again with the  following parameters, it works for me.
···
liorf:

  # Topics
  pointCloudTopic: "/os1_points"               # Point cloud data
  imuTopic: "/imu/data_raw"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "gps/fix"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useImuHeadingInitialization: true           # if using GPS data, set to "true"
  useGpsElevation: true                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 25.0                      # m^2, threshold for using GPS data
  
  # Export settings
  savePCD: false                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "/Downloads/LOAM/"        # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

  # Sensor Settings
  sensor: mulran                            # lidar sensor type, 'velodyne' or 'ouster' or 'livox' or 'robosense'
  N_SCAN: 64                                  # number of lidar channel (i.e., Velodyne/Ouster: 16, 32, 64, 128, Livox Horizon: 6)
  Horizon_SCAN: 1024                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048, Livox Horizon: 4000)
  downsampleRate: 2                           # default: 1. Downsample your data if too many points(line). i.e., 16 = 64 / 4, 16 = 16 / 1
  point_filter_num: 1                         # default: 3. Downsample your data if too many points(point). e.g., 16: 1, 32: 5, 64: 8
  lidarMinRange: 3.0                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 1000.0                       # default: 1000.0, maximum lidar range to be used

  # IMU Settings
  imuType: 0                                  # 0: 6-axis  1: 9-axis
  imuRate: 100.0
  imuAccNoise: 0.009939570888238808e-03
  imuGyrNoise: 0.005636343949698187e-03
  imuAccBiasN: 0.64356659353532566e-03
  imuGyrBiasN: 0.35640318696367613e-03
  imuGravity: 9.80511
  imuRPYWeight: 0.01

  # Extrinsics: T_lb (lidar -> imu)
  extrinsicTrans: [1.77, -0.00, -0.05]
  extrinsicRot: [-1, 0, 0,
                  0, -1, 0,
                  0, 0, 1]

  # This parameter is set only when the 9-axis IMU is used, but it must be a high-precision IMU. e.g. MTI-680
  extrinsicRPY: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]

  # voxel filter paprams
  mappingSurfLeafSize: 0.3                      # default: 0.4 - outdoor, 0.2 - indoor

  # robot motion constraint (in case you are using a 2D robot)
  z_tollerance: 1000                            # meters
  rotation_tollerance: 1000                     # radians

  # CPU Params
  numberOfCores: 4                              # number of cores for mapping optimization
  mappingProcessInterval: 0.0                  # seconds, regulate mapping frequency

  # Surrounding map
  surroundingkeyframeAddingDistThreshold: 1.0   # meters, regulate keyframe adding threshold
  surroundingkeyframeAddingAngleThreshold: 0.2  # radians, regulate keyframe adding threshold
  surroundingKeyframeDensity: 2.0               # meters, downsample surrounding keyframe poses   
  surroundingKeyframeSearchRadius: 50.0         # meters, within n meters scan-to-map optimization (when loop closure disabled)
  surroundingKeyframeMapLeafSize: 0.5           # downsample local map point cloud

  # Loop closure
  loopClosureEnableFlag: true
  loopClosureFrequency: 1.0                     # Hz, regulate loop closure constraint add frequency
  surroundingKeyframeSize: 50                   # submap size (when loop closure enabled)
  historyKeyframeSearchRadius: 15.0             # meters, key frame that is within n meters from current pose will be considerd for loop closure
  historyKeyframeSearchTimeDiff: 30.0           # seconds, key frame that is n seconds older will be considered for loop closure
  historyKeyframeSearchNum: 25                  # number of hostory key frames will be fused into a submap for loop closure
  loopClosureICPSurfLeafSize: 0.5               # downsample icp point cloud  
  historyKeyframeFitnessScore: 0.3              # icp threshold, the smaller the better alignment

  # Visualization
  globalMapVisualizationSearchRadius: 1000.0    # meters, global map visualization radius
  globalMapVisualizationPoseDensity: 10.0       # meters, global map visualization keyframe density
  globalMapVisualizationLeafSize: 1.0           # meters, global map visualization cloud density
···

v3loo, 2023-01-16:
I tried this configuration, and it works for much longer time and distance, but still failed at around second round of the routine. I also tested with downSample Rate set to 1 and got the same result.

I also tested on KITTI dataset which perfectly worked. Could this caused by the higher speed and larger map of MulRan?

YJZLuckyBoy, 2023-02-03:
> I tried this configuration, and it works for much longer time and distance, but still failed at around second round of the routine. I also tested with downSample Rate set to 1 and got the same [...]

Sorry for the late reply！I found that the IMU prediction on riverside01 and sejong02 is bad, may be this is caused by bad time sync between the lidar and IMU. 
You can change the following parameters and try again:
```
gpsTopic: "odometry/gpsz"

imuAccNoise: 0.009939570888238808e+02
imuGyrNoise: 0.005636343949698187e+02
imuAccBiasN: 0.64356659353532566e-03
imuGyrBiasN: 0.35640318696367613e-03

downsampleRate: 2
mappingSurfLeafSize: 0.3
```
Because of the z axis drift, the mapping result is not good.

shahaizi0, 2023-04-04:
> 我试过这个配置，它可以工作更长的时间和距离，但在例程的第二轮左右仍然失败。我还测试了将 downSample Rate 设置为 1 并得到了相同的结果。
> 
> 我还在 KITTI 数据集上进行了测试，效果非常好。这会不会是MulRan更高的速度和更大的地图造成的

我在kitty数据集上测试2011_09_30_drive_0028-004.bag会飘，你有什么建议吗