YJZLuckyBoy/liorf #7: compile error   libcommonlib.so incompatible

1574090787, 2023-01-29:
[ 89%] Linking CXX executable /home/nvidia/liorf-raw/devel/lib/liorf/liorf_imageProjection
/usr/bin/ld: 当搜索用于 /home/nvidia/liorf-raw/src/liorf-main/lib/libCommonLib.so 时跳过不兼容的 -lCommonLib
/usr/bin/ld: 当搜索用于 //usr/local/lib/libCommonLib.so 时跳过不兼容的 -lCommonLib
/usr/bin/ld: 找不到 -lCommonLib
collect2: error: ld returned 1 exit status
liorf-main/CMakeFiles/liorf_imageProjection.dir/build.make:768: recipe for target '/home/nvidia/liorf-raw/devel/lib/liorf/liorf_imageProjection' failed
make[2]: *** [/home/nvidia/liorf-raw/devel/lib/liorf/liorf_imageProjection] Error 1
CMakeFiles/Makefile2:2480: recipe for target 'liorf-main/CMakeFiles/liorf_imageProjection.dir/all' failed
make[1]: *** [liorf-main/CMakeFiles/liorf_imageProjection.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....

I use the arm kernel industrial computer

bclaw524, 2023-02-03:
I'm having the same issue. Using an Nvidia Jetson arm64 device.

YJZLuckyBoy, 2023-02-03:
> [ 89%] Linking CXX executable /home/nvidia/liorf-raw/devel/lib/liorf/liorf_imageProjection /usr/bin/ld: 当搜索用于 /home/nvidia/liorf-raw/src/liorf-main/lib/libCommonLib.so 时跳过不兼容的 -lCommonLib [...]

The ‘libCommonLib.so’ only depends on pcl library. Do you have the same problem with compiling lio-sam?

bclaw524, 2023-02-04:
> The ‘libCommonLib.so’ only depends on pcl library. Do you have the same problem with compiling lio-sam?

I'm able to compile lio-sam without issue. I have tried to edit the cmake file closer to the cmake I use for lio-sam and have managed to reach 100% but it still get the same error as the other post at the end and it doesn't compile.

alpogant, 2023-02-07:
I am trying to build on Jetson Xavier AGX with Ubuntu 20.04. LIO-SAM builds flawlessly but liorf giving this error: 
`/usr/bin/ld: skipping incompatible /home/user/liorf_ws/src/liorf/lib/libCommonLib.so when searching for -lCommonLib`
`/usr/bin/ld: cannot find -lCommonLib`
Is there an incompatibility with arm64 architecture processors?

YJZLuckyBoy, 2023-02-09:
> I'm able to compile lio-sam without issue. I have tried to edit the cmake file closer to the cmake I use for lio-sam and have managed to reach 100% but it still get the same error as the other post [...]

Hi! This problem is caused by the different compilation architectures of X86 and ARM. I have updated the code. You can clone the latest code for testing.

YJZLuckyBoy, 2023-02-09:
> Hi! This problem is caused by the different compilation architectures of X86 and ARM. I have updated the code. You can clone the latest code for testing.

Hi! This problem is caused by the different compilation architectures of x86 and ARM. I have updated the code. You can clone the latest code for testing.