YJZLuckyBoy/liorf #16: For frame [base_link]: Fixed Frame [map] does not exist

ParkerChan, 2023-05-09:
感谢您的开源！想请教一下，关于我使用liorf代码适配我自己的数据集的时候，rviz中一直显示没有map到base_link的tf，我运行的是lio_sam_default这个，我修改了对应yaml文件的数据话题、外参等，其他地方没有进行修改，是不是还有哪个地方需要我进行修改呢？

YJZLuckyBoy, 2023-06-26:
应该是frame_id设置不正确造成的，可以查看一下yaml文件中frame_id设置.

ParkerChan, 2023-06-26:
> 应该是frame_id设置不正确造成的，可以查看一下yaml文件中frame_id设置.

嗯嗯，感谢您的回复，我已经解决这个问题，并完美的运行了您的开源代码！再次感谢您的开源和回复！顺带再咨询一下，对于您的另一份开源代码liord_localization这个，如果我想实时的获取当前时刻的定位位姿信息，是不是直接从globalpath这个话题里面获取就行了，还有就是如果我后续想使用gnss的话，直接修改配置文件将gnss对应的地方设置为true就可以了是吗？谢谢您

YJZLuckyBoy, 2023-06-26:
liorf_localization定位结果可以使用"odometry/imu"，gnss数据满足要求的话直接修改就可以.

kristianhope, 2024-01-22:
Can you be more specific about how to fix the issue? I am experiencing the same problem, but cannot find a frame_id setting in lio_sam_ouster.yaml. I am using a 6-axis IMU (internal of Ouster OS1 128).

seajayshore, 2024-03-05:
I am also getting this problem and I can't figure-out what I'm supposed to change...

Are we supposed to make the lidar publish with a certain frame_id name?