YJZLuckyBoy/liorf #36: Robust initialization

kristianhope, 2024-04-17:
Hello. I am doing a thesis on lidar SLAM on an USV. I have successfully managed to run LIO-SAM (Liorf) in real-time with Ouster OS1-32 and its internal IMU. This works with and without GNSS fusion. I am now trying to increase the robustness. My biggest issue is initialization. Depending on what position and most importantly in what surroundings I start LIO-SAM I either start drifting really fast in one direction and point cloud blurs out, or I successfully initialize and can start moving the USV. If initialization is successful I can run it endlessly without problems. On the picture you can see a point cloud of the harbor area I am testing in. To give some scale the spaces between the piers/jetties are roughly around 40-50m wide. In the green circle areas, I can successfully initialize and start LIO-SAM, but in the red cross areas, and about anywhere else I am unable to initialize LIO-SAM. The second point cloud might show the area more clearly. This is the result of FAST-LIO2 with the same sensors, which is able to initialize pretty much anywhere. I have tried tuning the IMU noise and gravity (set to the local gravity at about 9.82), with no success. The main difference from starting in the green circled areas is that the USV is surrounded by features on all sides, unlike the red cross area where there is more open water (no features in the water obviously). My intuition tells me there must be some bias at initialization which cannot be corrected without sufficient features. I appreciate any input and help you might give me. Thanks in advance.

[image]

[image]

YJZLuckyBoy, 2024-04-29:
@kristianhope Fast-LIO2 is a tightly coupled solution, but LIO-SAM is loosely coupled. This may be due to insufficient point cloud data, which prevents normal mapping at the red position. You can change the following parameters and try again:
```
downsampleRate: 1
point_filter_num: 1
mappingSurfLeafSize: 0.15
```

kristianhope, 2024-05-01:
> @kristianhope Fast-LIO2 is a tightly coupled solution, but LIO-SAM is loosely coupled. This may be due to insufficient point cloud data, which prevents normal mapping at the red position. You can [...]

Can you elaborate on why you believe the tight-coupling of FAST-LIO2 enables it to initialise with fever points? I tried adjusting mappingSurfLeafSize, and I believed it improved a little bit. But still I need to be very close to land for it to work. Do you think it is caused by the lidar odometry or IMU-preintegration?