YJZLuckyBoy/liorf #40: 咨询一个RTK因子的问题 求助~   Ask a question about the RTK factor help!

lonlonago, 2024-05-24:
﻿
您好，我目前在建图过程中加入RTK的时候碰到了一些问题，然后我也阅读了您修改后的代码，但是还是有疑问，所以想请教一下，目前雷达和RTK之间是有一定的距离和旋转角度的，这个外参在哪里配置呢？
﻿
liosam 的原版本是通过修改launch/include/config/robot.urdf.xacro文件，“不然robot_localization功能包无法发布"odometry/gps"，修改此文件主要是添加imu->base_link，gps->base_link的tf变换，供robot_localization的节点查找并对IMU数据和GPS数据进行转换 ”，  这是您的博客写的，但是您修改后的liorf 已经不使用 robot_localization 了，那这个外参我应该在哪里，怎么配置呢？ 代码里面没有看到可以配置的地方。
﻿
另外就是我发现，如果我不配置这个外参，直接用这个RTK的数据，也可以建图，就是有2个问题，一是建图的时候，经纬度里程计和激光的里程计会有一个很大的方向偏差，大约90度，然后过了一会儿，激光里程计的方向会做一个很大角度的突然调整，然后后续就按照这个方向来走了，应该是激光的里程计的方向被RTK的强制纠正过来了，这很奇怪；  第二个问题是使用RTK的高度数据的时候，会出现激光里程计往上Z轴漂移，但是走着走着，隔了一段距离后，突然被RTK拉下来的情况，然后就导致地图出现了一个明显的断层，激光里程计也出现了一个明显的阶跃，请问这种情况该怎么处理，您有经验吗？

Hello, I have encountered some issues while adding RTK during the mapping process. I have also read your modified code, but I still have some questions. So, I would like to ask for advice. Currently, there is a certain distance and rotation angle between the radar and RTK. Where is this external parameter configured?

The original version of liosam was modified by modifying the launch/include/configure/robot. urdf. xacro file. Otherwise, the robot localization feature package cannot publish "odometry/gps". The main modification of this file is to add imu ->base_link, "The tf transformation of GPS ->base_link is used by robot_localization nodes to search for and convert IMU data and GPS data." This was written on your blog, but your modified Liorf no longer uses robot_localization. So where should I use this external parameter and how should I configure it?

Additionally, I have found that if I do not configure this external parameter and use the RTK data directly, I can still create a map. However, there are two issues. Firstly, during the mapping process, there is a significant directional deviation between the latitude and longitude odometers and the laser odometer, which is about 90 degrees. After a while, the direction of the laser odometer will be suddenly adjusted at a large angle, and then I will follow this direction. It is likely that the direction of the laser odometer has been forcibly corrected by the RTK, which is very strange; The second issue is that when using RTK altitude data, the laser odometer may drift up the Z-axis, but when walking for a distance, it is suddenly pulled down by RTK, resulting in a clear fault on the map and a clear step on the laser odometer. Do you have any experience in handling this situation?

lonlonago, 2024-05-29:
补充一下 我的雷达和IMU是放在一起的，但是RTK是另外单独放置的，可能和一般的配置不一样

kkkkkk812, 2025-04-24:
> 补充一下 我的雷达和IMU是放在一起的，但是RTK是另外单独放置的，可能和一般的配置不一样
我也遇到了相同的问题，请问这个问题您解决了吗

Pacemak7r, 2025-08-12:
我也碰到了这个情况，看了一天代码没找到怎么把RTK因子和LiDAR的因子统一到UTM坐标系下进行计算，最后建图感觉是被GPS因子强行拉正了不知道是不是gtsam对gps因子的强约束