YJZLuckyBoy/liorf #41: OS1-128 Internal IMU

CantLooseToAMoose, 2024-06-03:
When using the Ouster's internal IMU I get a `Large velocity, reset IMU-preintegration!` warning.
It also seems like the sensors position is jumping around when trying to construct the map.
How would I set it up to work with an internal IMU the right way? I used the `run_lio_sam_ouster.launch` file and did not change the config file.

ZhangYaoll, 2024-06-04:
Try to change imuRate from 500 to 100, because ouster imu rate is 100. Besides, extrinsicRot should change to [-1,0,0,0,-1,0,0,0,1]. Good luck.

CantLooseToAMoose, 2024-06-04:
> Try to change imuRate from 500 to 100, because ouster imu rate is 100. Besides, extrinsicRot should change to [-1,0,0,0,-1,0,0,0,1]. Good luck.

Sadly I tried that already..

CantLooseToAMoose, 2024-06-11:
Turns out my recorded bag file was from a really tight indoor space and I had to reduce the voxel size significantly lower to something like 0.1 for the algorithm to work.

closed 2024-06-11 by CantLooseToAMoose