YJZLuckyBoy/liorf #46: Point cloud is not in dense format, please remove NaN points first!

liu-ry, 2024-10-24:
使用 robosense 雷达的时候，报错“Point cloud is not in dense format, please remove NaN points first!”，请问是不是 robosense 雷达录制的数据暂时还不支持lio-sam算法？

wangxing124, 2025-02-25:
你好，请问你知道报错的原因了吗？我的是ouster的雷达也有这个报错

ylenianistico, 2025-03-04:
I have the same error with a Velodyne VLP-16. Do we need to first filer our data? (i don't speak chinese)

bboyack, 2025-04-22:
Not sure if anyone is still having this issue, but I fixed it by added simple code to just remove the NaNs instead of throwing the error.

In `imageProjection.cpp` in the `cachePointCloud` function, replace 
``` C++
        // check dense flag
        if (laserCloudIn->is_dense == false)
        {
            RCLCPP_ERROR_STREAM(get_logger(), "Point cloud is not in dense format, please remove NaN points first!");
            rclcpp::shutdown();
        }
```

with 
``` C++
        // check dense flag
        if (laserCloudIn->is_dense == false)
        {
            // Remove NaNs
            // RCLCPP_INFO(this->get_logger(), "Point cloud is not in dense format, removing NaN points");
            pcl::PointCloud<VelodynePointXYZIRT>::Ptr tmpCloud(new pcl::PointCloud<VelodynePointXYZIRT>);
            std::vector<int> indices;
            pcl::removeNaNFromPointCloud(*laserCloudIn, *tmpCloud, indices);
            laserCloudIn = tmpCloud;
        }
```